betaflight/betaflight.com #121: docs/wiki/firmware-flasher suggestions

jawz101, 2023-01-25:
https://github.com/betaflight/betaflight.com/blob/master/docs/wiki/configurator/firmware-flasher-tab.mdx

Under the "Other Options" section, can it be more descriptive about what DASHBOARD means?  Are we saying that you will not get an on-screen display unless you build it into the ROM?  Why would that not be a default option?

`ACRO TRAINER` custom Define should be `ACRO_TRAINER`

haslinghuis, 2023-01-26:
`DASHBOARD` depends on `OSD` and is providing an overall information page.

With firmware size overflowing we decided to provide a hardware base (core) build with features on top.
Please pick the required features which would not be available before.

jawz101, 2023-01-27:
Is it the screen that comes up at the end of a flight when you disarm?

haslinghuis, 2023-01-27:
No it's not the statistics screen :) - it's more of an overview screen

jawz101, 2023-01-27:
lol ah.  `Post-Flight overview screen`.  I couldn't figure out how to word it.  But yeah, could the documenation say something like that?

Option | Description
-- | --
AKK (SA Fix) | Smartaudio patch for AKK hardware
DASHBOARD | OSD Dashboard
DASHBOARD | Post-flight Overview Screen

jawz101, 2023-01-27:
Thanks.  I just submitted a pull request.  I didn't realize I could do that

closed 2023-01-27 by jawz101