betaflight/betaflight.com #194: [Request] List for tested GNSS devices to be added as supported in documentation

haslinghuis, 2023-06-17:
Would be good practice to create a list for supported and tested GNSS devices.

List should entail manufacturer, model, type, ublox protocol version and maybe some other attributes.

Starting the list here we can accumulate some data before adding to the docs.

haslinghuis, 2023-07-05:
Starting from here: https://github.com/betaflight/betaflight-configurator/issues/3478?notification_referrer_id=NT_kwDOAH9U_rI2NzE5OTI2NzA3OjgzNDQ4MzA#issuecomment-1622429046

FOXEER M10Q-180
https://www.foxeer.com/foxeer-m10q-180-gps-5883-compass-g-507

MATEKSYS M10Q-5883
http://www.mateksys.com/?portfolio=m10q-5883

NEEWBEEDRONE M10 Micro with compass
https://newbeedrone.com/collections/all/products/newbeedrone-m10q-micro-gps-module-with-compass

FLYWOO GOKU GM10 Nano V3
https://flywoo.net/products/goku-gm10-nano-v3-gps-w-compass

HGLRC GPS-M100 MINI
https://www.hglrc.com/products/hglrc-m100_mini-gps

A-SKY AK413 ubx M10050-KB
https://www.aliexpress.com/i/3256805181303434.html?gatewayAdapt=4itemAdapt

Pancronos, 2023-08-12:
walksnail WS-M181 (COPJ-18GPS)
site: https://caddxfpv.com/products/walksnail-ws-m181gps
foxeer m10q-120
site: https://www.foxeer.com/foxeer-m10q-120-gps-5883-compass-g-517

haslinghuis, 2023-08-13:
[M10_GPS_tested_in_BF.xlsx](https://github.com/betaflight/betaflight.com/files/12329617/M10_GPS_tested_in_BF.xlsx)

Thanks Mat !

Pancronos, 2023-08-13:
No worries. I will see if i can fill for those missing it, the standard orientation. At least on the bench.

YarosMallorca, 2024-07-06:
@haslinghuis I tested GepRC M10 GPS (GEP-M10-DQ) in BF 4.4 and 4.5, and it works correctly.

patrikpordi, 2025-11-20:
ARK SAM GPS: https://arkelectron.com/product/ark-sam-gps/