betaflight/betaflight.com #197: Supported MCUs

woutput, 2023-07-07:
Hi great Betaflight community!

While choosing parts for my next build, I found some information regarding the MicroController Unit (MCU) on the Flight Controller (FC) that is unclear to me.

1. On [this page on manufacturer design guidelines](https://betaflight.com/docs/manufacturer/manufacturer-design-guidelines) the MCU column of the table in paragraph 4.1 lists "F411", "F405" and "F7X2, H7XX, G4XX, and similar" which are part of "... **strongly recommended** default configurations."
   - This appears to imply that the STM32F745VGT6 _would not be supported_. Is that device supported? If so (if we want to keep this table (see later in this post)) then this could be updated to be more clear.
   - Also: many cells under the MCU column are empty (if we want to keep this table (see later in this post)), I propose to fill all cells under the MCU column to be clear.
2. [This page](https://betaflight.com/docs/manufacturer/creating-an-unified-target) on creating [_a_](https://www.scribbr.com/commonly-confused-words/a-vs-an/) unified target provides another list of MCUs in paragraph 2.8. It is equal to the list of assets on [the releases page](https://github.com/betaflight/betaflight/releases).
   - It includes STM32F745 which implies that the STM32F745VGT6 _is_ supported
   - Is STM32F765VIT6 supported? Why (not)?
3. Flash and RAM size. From what I could find, these devices have 512-2048 KiB of flash memory to store the Betaflight firmware (compiled code + data) and they have 128-1024 KiB of RAM to run it. Already in June 2018, the [release notes of v3.4.0-RC1](https://github.com/betaflight/betaflight/releases/tag/v3.4.0-RC1) contained the text "unfortunately bugfixes and improvements in the flight controller core functionality have led to an increase of the firmware size, causing it to overflow the available space on a number of F3 based flight controllers. As a result, some features have had to be removed from a number of F3 based flight controllers in order to _make the firmware fit into **flash**_" Does this mean that the 256 KiB of flash in the STM32F303CCT6 became insufficient? Was and/or is the 40 KiB of RAM sufficient? Today, [the release notes](https://github.com/betaflight/betaflight/releases/tag/4.4.0) contain a similar message: "4.4 has a new Cloud Building System. This is design to extend the life of smaller flash sized MCUs". And what about the future? (also see [here](https://intofpv.com/t-betaflight-too-big) and [here](https://www.youtube.com/watch?v=YmTq68M-NkU) and [here](https://github.com/betaflight/betaflight-configurator/issues/3315))
4. F722. From what I could find, some FCs use the STM32F722RET6 (with 512 KiB of flash) while others use the STM32F722RGT6 (with 1024 KiB of flash). In other cases, it is unclear which of the two (or even third?) is on the FC. Today, it looks like Betaflight assumes that F722 means STM32F722RET6 because of the comments in [this code](https://github.com/betaflight/betaflight/blob/master/src/link/stm32_flash_f722.ld) and the statement "... smaller flash sized MCUs (F411 and F722) ..." in [the release notes of 4.4.0](https://github.com/betaflight/betaflight/releases/tag/4.4.0)
   - Example 1: [this stack](https://www.getfpv.com/electronics/stacks/skystars-fly-stack-f7-hd-mini-pro-fc-km55-55a-am-32-3-6s-4-in-1-esc-20x20.html) states in the text that a STM32F722RGT6 is used while if you zoom in on [the chip in the picture](https://cdn.getfpv.com/media/wysiwyg/Stacks/F7miniHDpro-spec-scaled_copy.webp), it shows STM32F722RET6
   - Example 2: [this stack](https://www.speedybee.com/speedybee-f7-v3-bl32-50a-30x30-stack/#Parameters) lists "STM32F722" as MCU, also in [the manual](https://store-fhxxhuiq8q.mybigcommerce.com/product_images/img_SpeedyBee%20F7V3_STACK/SpeedyBee_F7V3_Stack_Manua1.2_EN.pdf). Also no MCU flash or RAM size are described.

To solve/improve this, I suggest to have one single table that
- lists complete part numbers (like STM32F722RET6 and STM32F722RGT6 instead of only partial ones like STM32F722) to be specific
- lists flash and RAM size to help buyers decide
- lists first and last Betaflight version that supports the device
- is the single source of truth for past, current and future Betaflight releases (existing location(s) mentioned under 1. and/or 2. shall refer to a single table)
- provides advice to manufacturers (maybe try to minimize power usage, be a STM32 device, have at least X MHz clock, be a Cortex-M? core, have at least X KiB flash, have at least X KiB RAM, etc.)
- provides advice to drone designers, maybe use similar terminology as [this example](https://www.analog.com/en/support/customer-service-resources/sales/product-life-cycle-information.html)

Maybe this can be used as starting point:

| Type          | Advice                          | ARM Cortex M# | Clock [MHz] | Flash [KiB] | RAM [KiB] | UARTs |
|---------------|---------------------------------|---------------|-------------|-------------|-----------|-------|
| STM32F303CCT6 | Obsolete                        | 4             | 72          | 256         | 40        | 3     |
| STM32F405RGT6 | Not recommended for new designs | 4             | 168         | 1024        | 192       | 5     |
| STM32F411CEU6 | Not recommended for new designs | 4             | 100         | 512         | 128       | 2     |
| STM32F722RET6 | Not recommended for new designs | 7             | 216         | 512         | 256       | 5     |
| STM32F722RGT6 | Recommended for new designs     | 7             | 216         | 1024        | 276       | 6     |
| STM32F745VGT6 | Recommended for new designs     | 7             | 216         | 1024        | 340       | 6     |
| STM32F765VIT6 | Obsolete                        | 7             | 216         | 2048        | 512       | 4     |
| STM32H743VIT6 | Recommended for new designs     | 7             | 480         | 2048        | 1024      | 7     |
| STM32H743VIH6 | Recommended for new designs     | 7             | 480         | 2048        | 1024      | 7     |

Please note that this table would need to contain reliable information; I used multiple more or less reliable sources to create this; I recommend to add references to reliable resources for the final version.

**What do you think?**

P.S. Please do let me know if I can help in any way; split this issue into multiple issues? mention this issue in another repo? create a pull request? etc.

P.P.S.
Just a random but big THANK YOU to @borisbstyle and @haslinghuis because of all their great work for Betaflight!

sugaarK, 2023-08-10:
@woutput thanks for this lets see if we can get it into the hardware doc

woutput, 2023-10-10:
> @woutput thanks for this lets see if we can get it into the hardware doc

You're very welcome.
What do you suggest exactly?
Thanks

haslinghuis, 2023-10-11:
@woutput we need to add some other MCU's we have added lately, appreciate your work.

ctzsnooze, 2023-12-20:
@woutput are you able to raise PR's on GitHub?

haslinghuis, 2023-12-20:
Adding them shortly

woutput, 2023-12-20:
> @woutput are you able to raise PR's on GitHub?

Sure. Let me know how I can help.

haslinghuis, 2023-12-20:
If you could raise a PR adding this info would help a lot. Have kept pushing it forward all the time because of time restrictions / prioritizing over firmware and client software.

woutput, 2024-01-20:
Okay, let me take a look...

nerdCopter, 2024-02-20:
i set a "Pinned" label so that it does not auto-close.  Betaflight is open to Pull-Requests.