betaflight/betaflight.com #211: A little blur with light theme

McGiverGim, 2023-08-04:
Hi! The light theme seems to me that makes some texts a little blur. I don't know sure why, I suppose is the effect of the contrast with the background.

[image]

I'm the only one that thinks that is a little blur?

haslinghuis, 2023-08-04:
On linux:
[image]

McGiverGim, 2023-08-04:
Where is the background? 😱

haslinghuis, 2023-08-04:
¯\_(ツ)_/¯
 
 Same on windows (edge)

haslinghuis, 2023-08-04:
Firefox on Windows:

[image]

haslinghuis, 2023-08-04:
[image]

haslinghuis, 2023-08-04:
Only have 1366x768 and 1600x900 devices with me here for testing.

McGiverGim, 2023-09-04:
I'm out of home. I will keep it open until I return and can do more tests.

nerdCopter, 2023-10-05:
have not looked deeply, have to play with these things:
[image]

increasing blur px helped, but is the wrong solution. i feel like opacity might help, but do not immediately know where that "var" is set.

McGiverGim, 2023-11-17:
The issue remains, but simply I don't know how to fix it.

nerdCopter, 2023-11-17:
@VitroidFPV @freasy

McGiverGim, 2023-12-18:
Keep on!

VitroidFPV, 2023-12-19:
Will have a look

ctzsnooze, 2023-12-19:
This is the Download page on Mac with Chrome.
Only Development and Download have the fancy background.
All the text is sharp.

 'You can find release planning here' is grey, and lacks contrast with the pale grey background.
 
 My main frustration is the lack of contrast between the orange text and grey backgrounds in daylight mode.  It's just too difficult to read. We probably need a darker orange, eg #c90.  This in HSL is a color angle of 45 degrees and luminance of 50.  I think our stock orange is #fc3 which is the same color angle, but luminance of 60..

ctzsnooze, 2023-12-20:
@McGiverGim is it improved a little bit now?

McGiverGim, 2023-12-21:
Yes it seems better now
[image]

I'm not sure if it's because the colours have changed or simply because the shade is at different position than in my first screen capture. But at this moment the issue is fixed for me.

haslinghuis, 2023-12-21:
We also had missing commas in color definition for 750: 'rgb(44 44 44)',

Found in https://github.com/betaflight/betaflight.com/pull/288

closed 2024-01-29 by github-actions[bot]