betaflight/betaflight.com #224: Board support notices

khari05, 2023-09-06:
Hi there, I just wanted to let you know that the current wiki still includes the STM32F3Discovery as a supported board target. I understand that it is technically supported but you might want to add a deprecation/old firmware warning to the STM32 dev board page for the F3 board.

haslinghuis, 2023-10-14:
Agree the board sections needs updating as we already use a board link in configurator for easy access.

fbdfbd3371, 2024-01-10:
Good afternoon! Please tell me why the mpu6000 gyroscope is listed in the wiki on board tmotorf7. The website https://store.tmotor.com/product/f7-flight-controller.html lists the BMI270 gyroscope. Thanks a lot!

nerdCopter, 2024-01-10:
> Good afternoon! Please tell me why the mpu6000 gyroscope is listed in the wiki on board tmotorf7. The website [store.tmotor.com/product/f7-flight-controller.html](https://store.tmotor.com/product/f7-flight-controller.html) lists the BMI270 gyroscope. Thanks a lot!

Generally, hardware manufacturers change gyro chips at will without regard to software that supports them.
`TMOTORF7` has had several versions with the following gyros: `BMI270`, `ICM42688P`, `MPU6000`, `MPU6500`.

closed 2024-02-17 by github-actions[bot]