betaflight/betaflight.com #295: Disable Hardware ADC filter

fubz, 2023-12-21:
### Describe the bug

Betaflight receiver says Disable the hardware ADC filter in the transmitter if using OpenTX or EdgeTX.  There is a link that takes me to https://github.com/betaflight/betaflight/blob/master/docs/Rx.md#disabling-the-opentxedgetx-adc-filter which is a 404

### To Reproduce

Click the link https://github.com/betaflight/betaflight/blob/master/docs/Rx.md#disabling-the-opentxedgetx-adc-filter 
See a not real page.

### Expected behavior

Click the link and be directed towards the helpful information required to setup the quad.

### Support ID

```txt
*** Support data submitted *** 
Id: "b6f8d3fe-8200-47be-8af7-2c86d2bafcbb"
```

### Flight controller

MTKS/MATEKF405

### Other components

Taranis X9D

haslinghuis, 2023-12-21:
Yes sorry - documentation is being restructured.

Can you tell where you clicked the link? Probably configurator?

haslinghuis, 2023-12-21:
When clicking link in configurator we get https://betaflight.com/docs/development/Rx#disabling-the-opentxedgetx-adc-filter so this might already be solved?

haslinghuis, 2023-12-21:
Closing as resolved.

closed 2023-12-21 by haslinghuis