betaflight/betaflight.com #389: make TARGET=MATEKH743 recommended in documentation doesn't work

valeriyvan, 2024-03-07:
Please refer to issue https://github.com/betaflight/betaflight/issues/13425.

Neither `make CONFIG=MATEKH743` nor `make MATEKH743` is mentioned in official documentation on page https://betaflight.com/docs/development/building/Building-in-Mac-OSX. But it twice mentions `make TARGET=MATEKH743`. What's correct, canonical way of building,  `make CONFIG=MATEKH743` or `make MATEKH743`? I would like to fix documentation.

Here I mean specifically build instructions for macOS. But I suppose same apply for other platforms as well. Build instructions for other platforms recommend to use `make CONFIG=MATEKH743` as well.

For what should I change `make CONFIG=MATEKH743` in documentation, `make CONFIG=MATEKH743` or `make MATEKH743`?

valeriyvan, 2024-03-07:
I got it. I should update documentation giving examples of building one of unified targets instead of MATEKH743.

nerdCopter, 2024-03-07:
Until documentation is complete, i would suggest checking the "log" of a Configurator cloud-build, which will report a proper command structure.
[image]
[image]
be sure to `make configs` to update the targets first.

nerdCopter, 2024-03-07:
I believe the above cloud-build commands would be the most proper for pilots.

However, to assist...

`make TARGET=` is only for the MCU type. e.g. make `TARGET=STM32H743`. where you can then load a specific `config.h` when flashing.

I believe the remaining make commands will produce targets with all features/drivers unless i am mistaken.  I am unsure how to produce a "core-only" from commandline. Maybe @haslinghuis can correct me or add information.

Interestingly i found that the `make MATEKH743`  vs `make CONFIG=MATEKH743` produced binaries that were dissimilar in some capacity:
```diff
--- ./betaflight_4.5.0_STM32H743_MATEKH743_config.hex	2024-03-07 07:56:04.432910861 -0600
+++ ./betaflight_4.5.0_STM32H743_MATEKH743_make.hex	2024-03-07 07:54:48.615961196 -0600
@@ -29850,7 +29850,7 @@
 :104660007329202573202573202F202573202825CA
 :104670007329204D5350204150493A202573003171
 :104680002E3436003664393864626237340030378D
-:104690003A35353A3233004D6172202037203230BE
+:104690003A35343A3033004D6172202037203230C1
 :1046A000323400342E352E300033643931386339DA
 :1046B000002320636F6E666967207265763A202555
 :1046C00073004D4154454B48373433004D544B53E0
@@ -30315,7 +30315,7 @@
 :1063700000000000434F4E46494755524544000037
 :106380004254464C0000000053544D3332483734D9
 :10639000330000004D61722020372032303234004B
-:1063A00030373A35353A3233000000003664393838
+:1063A00030373A35343A303300000000366439383B
 :1063B00064626237340000000A50323C3D3E3F4088
 :1063C00041424344454647000A323C3D3E3F40413E
 :1063D0004243444546470000484F52495A4F4E00F9
```

haslinghuis, 2024-03-07:
`make TARGET=BOARD` was used before 4.4

`make CONFIG=BOARD` was introduced here: https://github.com/betaflight/betaflight/pull/12341 for 4.4 specifically IIRC as it had special use case while transitioning from unified targets to config repo.

For 4.5 we need `make configs` and 
- `make BOARD`
- `make BOARD -EXTRA_FLAGS="-DCLOUD_BUILD"` --> needs to specify other includes (see cloud build log)
- `make BOARD -EXTRA_FLAGS="-DCORE_BUILD"`   --> only includes drivers - no features.

haslinghuis, 2024-03-12:
Was attended by @hydra about the typo. `EXTRA_FLAGS` does not need a minus sign in front, 
i.e. the command should be 

`make BOARD EXTRA_FLAGS="-DCLOUD_BUILD"`

closed 2024-04-22 by github-actions[bot]

keu, 2025-01-05:
Any updates on this? 
It is very frustrating to have no single source of truth as of today. Neither the README nor the official documentation mentions the correct command to build a single target

blckmn, 2025-01-05:
> Any updates on this? It is very frustrating to have no single source of truth as of today. Neither the README nor the official documentation mentions the correct command to build a single target

`make MATEKH743` to make the specific configuration. Make will tell you if you need to hydrate the configs first with `make configs` or not.

`make TARGET=STM32H743` is used if you want a generic target for a particular MCU type (aka platform), but without the configuration defaulted to any particular board setup (you would need to use CLI to even get gyro setup).